kolins-cz/Smart-BMS-Bluetooth-ESP32 #2: could the code be used to send the data to MQTT server

iz3man, 2021-02-19:
Not an issue at all, but I'm looking to integrating a LifePo4 battery from a solar system into my home automation. This could be most easily done by sending the data gathered by the ESP32 to and MQTT server.
Any idea if this could be done easily? Unfortunately I'm no coder at all ;)
Thanks in advance!

kolins-cz, 2021-02-19:
It should be easy. Just include some working mqtt library and send data to mqtt server in main loop. All data should be in packBasicInfo and packCellInfo variables.
I don't need mqtt for my ebike so I will not do it, sorry. But it should be easy enough. Maybe in a few years when I got my solar setup.

closed 2021-02-19 by kolins-cz

iz3man, 2021-02-19:
LOL. MQTT on a bike is useless - I agree :)
I just went throught the code a little bit. Should be doable. Thanks!

iz3man, 2021-02-20:
Maybe you still want to support a little here. If not, or this is not the correct place just let me know and I'll stop.
I now integrated Wifi and Mqtt, and can send data to my server.

Unfortunately mqqt publish only sends STRING, and you of course are using FLOAT for the read values.

```
    packBasicInfo.Volts = packBasicInfo.Volts / 1000;
    dtostrf(packBasicInfo.Volts, 6, 2, STRpackBasicInfoVolts);
    client.publish("Total voltage", STRpackBasicInfoVolts);
```

If I do so I only see 52.000 Volts, where it should be something like 52.123V. Any idea why?

And this code here:

`    commSerial.printf("Number of cells: %u\n", packCellInfo.NumOfCells);
    for (byte i = 1; i <= packCellInfo.NumOfCells; i++)
    {
        commSerial.printf("Cell no. %u", i);
        commSerial.printf("   %f\n", (float)packCellInfo.CellVolt[i - 1] / 1000);
    }`

I need to translate all those single voltage values to STRING, but got no idea how. Can you please help?

Thanks

kolins-cz, 2021-02-21:
Hi,
I do not understand why it does show only 52.000V. On my ebike it shows battery voltage properly.  As a matter of fact I have not touched the code since posting it here on github. I believe there is sprintf or another function in C or C++ to do this. Or maybe String class should be able to do it. I have programmed this out of necessity. I took me a while to do it and a lot of help from my colleagues. I am more of hardware guy. Sorry.