kolins-cz/Smart-BMS-Bluetooth-ESP32 #4: Ant-bms is compatible? ANT-BLE16S device name

rdaguerre, 2021-02-22:
Hi, I've been looking for information on ant-bms for a long time. I am needing to use the BLE to use with the micropython and esp32. The only thing I could get so far is to use the command 0xdbdb00000000 and send me 20bytes in response. In other ble app i can read few more bytes. Could you tell me if these commands work here? Because I tried a couple of status from here and I don't get any notification. Greetings and I hope someone knows. Greetings

kolins-cz, 2021-05-23:
Sorry, I have no idea. But I will leave this issue open and hopefully someone will answer you.

lucasimons, 2021-09-20:
I have found this repo that looks amazing there is any updates?

Maxoos, 2021-12-06:
@rdaguerre I'm trying to work it out myself with ESP32 and BLE. As far as I understand, you subscribe to the service and then write {0xDB, 0xDB, 0x00, 0x00, 0x00, 0x00}. You will get in the callback a series of bytes, it looks like this:

```
18:40:52.317 -> data: 0xAA 55 AA FF 02 DC 0E 4F 0E 52 0E 4F 0E 50 0E 50 0E 53 0E 4F Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 20
18:40:52.350 -> data: 0x0E 4F 0E 50 0E 52 0E 4F 0E 52 0E 51 0E 4F 0E 50 0E 4E 0E 4E Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 20
18:40:52.350 -> data: 0x0E 50 0E 51 0E 4E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 20
18:40:52.350 -> data: 0x00 00 00 00 00 00 00 00 00 00 00 00 00 00 30 03 0A 32 C0 01 Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 20
18:40:52.384 -> data: 0x76 1C 83 00 01 68 0B 00 27 E9 59 00 1A 00 1C 00 17 00 17 FF Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 20
18:40:52.384 -> data: 0xD8 FF D8 01 01 00 00 00 00 00 0F 00 00 27 0F 06 0E 53 10 0E Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 20
18:40:52.422 -> data: 0x4E 0E 4F 14 00 00 00 82 00 7E 02 AB 00 00 00 00 00 00 13 A1 Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 20
```

I'm now trying to unpack this and understand how it work, but I think you just need to add them all together. It just sends 7 notifications 20 bytes each so it's 140 bytes together.

gmbo, 2021-12-06:
Look here: [user manual.zip](https://drive.google.com/file/d/1E0HEes6kgBze5DI5MRO2mkE8BmNdzXas/view)
There is a good, complete description.
With this it should be possible to decode it.
Next to the BMS user manual there is a subfolder with the communication protocols.

Maxoos, 2021-12-09:
@gmbo Brilliant! thanks!