kolins-cz/Smart-BMS-Bluetooth-ESP32 #5: pinout not ttgo

cybermandiant, 2021-05-08:
hello for st7735 and esp32 what pin for plug st7735 ???or ili9341 2.4
esp32 lite 
thanks

gmbo, 2021-12-06:
If I understand your short request correctly you have problems to get a SPI display running with the ttf_eSPI library.
I had too, until I understood the user_setup.
You have to change it in the directory of the lib or copy it there.
it is good to keep a copy outside the lib, because when updating the lib it will be overwritten.
With esp lite you mean something like this [Lolin Lite](https://cdn.shopify.com/s/files/1/1509/1638/files/ESP32_Lolin32_Pinout.pdf?v=1608455156) I guess.
The SPI pins are default in the normal case. The CS you can assign freely. 
You still need an output for the backlight, and if you use it the pins for the touch field.

Special pins that are available on the ESP32:
GPIO0: enable flash mode
GPIO1/3: for UART/Serial
GPIO2: must not be pulled HIGH at boot time
GPIO12: ESP32 will not boot if this GPIO is pulled HIGH at boot time

GPIO34, 35, 36 (VP), 39 (VN): only inputs GPIs, and have no internal pullup resistors.

I myself use an [ArduiTouch with an ESP32DevC](https://www.hwhardsoft.de/app/download/11468539097/ArduiTouch+ESP+schematic+V01-03-00.pdf?t=1619450789) .
that works smoothly.
It should work the same way with an ESP32 Lite.
This would be my User_Setup.h 

```

#define ILI9341_DRIVER       // Generic driver for common displays
//#define ST7735_DRIVER      // Define additional parameters below for this display
#define TFT_BL   15            // LED back-light control pin
#define TFT_BACKLIGHT_ON LOW  // Level to turn ON back-light (HIGH or LOW)
#define TFT_BACKLIGHT_OFF HIGH  // Level to turn ON back-light (HIGH or LOW)
// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5  // Chip select control pin
#define TFT_DC    4  // Data Command control pin
#define TFT_RST  22  // Reset pin (could connect to RST pin)

#define TOUCH_CS 14     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT
#define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  55000000 // STM32 SPI1 only (SPI2 maximum is 27MHz)
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

```
Only pin 22 could make difficulties, because there is the buildin LED.
And the line 
`#define TFT_BACKLIGHT_ON LOW  // Level to turn ON back-light (HIGH or LOW) `

had to be changed from HIGH to LOW, because the transistor T1 inverts the signal.

I hope this helps you.

cybermandiant, 2021-12-06:
Excelent thanks a lot i test this during this week ;)