kolins-cz/Smart-BMS-Bluetooth-ESP32 #7: No Basic data is reported.

Paulf007, 2022-05-17:
I managed to get a connection and the code reports the voltage for each individual cell but for some reason I cannot het the basic data
eg : Volts , Amps , Capacity Remain Ah - they all show 0.000 value.
Is there a way to verify that the data is accepted / incoming ?

gmbo, 2022-05-17:
Es sieht so aus als wenn Sie nur die Info4 empangen. bmsGetInfo4(); ist die Aufforderung an den Akku das zu senden.
Normalerweise wird getoggled zwischen  bmsGetInfo3(); und  bmsGetInfo4(); beide senden eine Info an den seriellen Monitor.
Danach sollte ein Paket empangen werden und nicht  "Invalid packer received" melden.
Die Auswertung der Info3 in processBasicInfo(..) fragt nur die Länge auf 0x1B ab und könnte bei einem etwas abgewandelten Akku fehlschlagen. In den Start der Routine könnte man ein commSerial.printf("Datenlänge: %x\n", dataLen); einfügen um das zu kontrollieren.
It looks like you only receive the **Info4. bmsGetInfo4();** is the request to the battery to send that.
Normally getoggled between **bmsGetInfo3();** and **bmsGetInfo4();** both send an info to the serial monitor.
After that a packet should be received and not report "Invalid packer received".
The evaluation of Info3 in **processBasicInfo(..)** only queries the length on **0x1B** and might fail with a slightly modified accumulator. In the start of the routine you could insert a **commSerial.printf("Data length: %x\n", dataLen);** to check this.

Translated with www.DeepL.com/Translator (free version)

Paulf007, 2022-05-17:
You were correct , thank you. I also assumed as much as I commented out the compared value. The length was 0X1D. 
Can there be a way that one could allow for deviations. I currently added the following to the code (Below)

Sie hatten Recht, danke. Ich bin auch davon ausgegangen, als ich den Vergleichswert auskommentiert habe. Die Länge war 0X1D. 
Kann es eine Möglichkeit geben, dass man Abweichungen berücksichtigen kann. Ich habe gerade folgendes in den Code eingefügt 
  

`  if (dataLen != 0x1D)
  {
    return false;
  }`

Translated with [www.DeepL.com/Translator](http://www.deepl.com/Translator) (free version)

gmbo, 2022-05-17:
Es sollte dann aber ein anderer Typ der Batterie sein und damit auch andere Daten um was für eine Batterie handelt es sich in dem Fall und was ist in dem zusätzlichem Byte an Wert enthalten.
Von daher müsste 
`if ((dataLen != 0x1B)&&((dataLen != 0x1D)) { return false; }`
richtiger sein und dann in der Auswertung ` If (dataLen == 0x1D) {setze Zusatzbeit ... }`
But then it should be a different type of battery and therefore also different data what kind of battery it is in this case and what is contained in the additional byte of value.
Therefore it should be 
`if ((dataLen != 0x1B)&&((dataLen != 0x1D)) { return false; }`
should be more correct and then in the evaluation `If (dataLen == 0x1D) {set additional post ... }`

Paulf007, 2022-05-18:
Ich werde in einigen Foren nachsehen müssen, was in diesem Byte ist.
Bis jetzt laufen die Programme nach der Änderung gut. 
Was wird passieren, wenn ich die Bedingung vollständig zu entfernen? 

I will need to check on some of the forums what in that byte is.
So far the programs is running well after the change. 
What will happen if I remove the condition completely?

gmbo, 2022-05-18:
Das ist nur eine Sicherheitsabfrage und es wäre ja möglich dass die Zuordnung in dem Telegramm eine andere ist.
Daher hätte ich beides zugelassen um dann später die Differenzen zu bearbeiten. ich bin mir nicht sicher ob nicht ein & reicht.
This is only a security query and it would be possible that the assignment in the telegram is different.
Therefore, I would have allowed both to then later edit the differences. I'm not sure if not one & is enough.

bigmacblau, 2023-01-25:
Hallo zusammen, ich stehe vor dem selben Problem das die Basisdaten leer d.h. =0 sind. Die Zelldaten sind jedoch vorhanden. Alle Dokumentationen dazu sind gleich. Es muss aber eine andere Datenstruktur zurück gegeben werden "Invalid packer received"  die das Programm nicht verarbeiten kann. Gibt es dafür eine Lösung?
Vielen Dank
Wolfgang

gmbo, 2023-01-25:
Hallo Wolfgang, hast du dir mal ausgeben lassen was empfangen wird?  Ich würde mir dafür in BMS_process_data.ino der Funktion  isPacketValid(() einige Ausgaben machen lassen.
```

bool isPacketValid(byte *packet) //check if packet is valid
{
  TRACE;
  if (packet == nullptr)
  {
    commSerial.println("oh nullptr");
    return false;
  }

  bmsPacketHeaderStruct *pHeader = (bmsPacketHeaderStruct *)packet;
  int checksumLen = pHeader->dataLen + 2; // status + data len + data

  if (pHeader->start != 0xDD)
  {
   commSerial.printf("Startheader nicht 0xDD sondern: %x\n", pHeader->start);
    return false;
  }

  int offset = 2; // header 0xDD and command type are skipped

  uint16_t checksum = 0;
  for (int i = 0; i < checksumLen; i++)
  {
    checksum += packet[offset + i];
  }

  commSerial.printf("berechnete Summe: %x\n", checksum);

  checksum = ((checksum ^ 0xFFFF) + 1) & 0xFFFF;
  commSerial.printf("checksum gerechnet: %x\n", checksum);

  uint16_t rxChecksum = two_ints_into16(packet[offset + checksumLen], packet[offset + checksumLen + 1]);

    commSerial.println(packet[offset + checksumLen],HEX);
    commSerial.println(packet[offset + checksumLen + 1],HEX);
    commSerial.println(packet[offset + checksumLen + 2],HEX);

  commSerial.printf("empfangene checksumme: %x\n", checksum);

  if (checksum == rxChecksum)
  {
    commSerial.printf("Packet is valid\n");
    return true;
  }
  else
  {
    commSerial.printf("Packet is not valid\n");
    commSerial.printf("Expected value: %x\n", rxChecksum);
    return false;
  }
}

```
Dann solltest du mehr erfahren.

bigmacblau, 2023-01-27:
Hallo gmbo,
danke für Deine Hilfe. Ich habe deinen Code eingefügt und bekomme jetzt gültige und ungültige Checksummen angezeigt.
Jedoch, was fange ich damit an? Du sieht ich bin in der Arduino Welt nicht zuhause. ;-)  Kannst Du mir weiterhelfen?
Danke 
Wolfgang
PS: gibt es eine Möglichkeit die Einzel Cellspannungen, die ja angezeigt werden, über MQTT zu veröffentlichen, das würde mir nämlich schon reichen?

gmbo, 2023-01-28:
Hallo Wofgang, 
ich kann versuchen dir zu helfen.
Mqtt sollte nicht das Problem sein, das geht auf jeden Fall.Hast du eine Vorstellung wie dein payload aussehen soll?
Ich sende da zur Zeit nur Daten aus dem Basic Telegramm, mit vielen einzelnen Topics, würde ich heute wohl in eier Json Struktur machen, oder eben über LoraWAN.
Trotzdem gehe ich davon aus, dass es von Vorteil wäre, den eigentlichen Fehler zu finden. 
Immerhin sollten ja die Telegramme reinkommen und die Anfangskennung 0xDD ist vorhanden.
Ich schaue morgen mal wo wir das Telegramm ausgeben können um zu sehen warum da etwas falsch gerechnet wird.
Was für einen Akku ließt du aus und hast du darauf immer Zugriff. 
Was hast du als ESP für ein Bord und welche Art von Anzeige hast du daran?
LG
Gisbert

bigmacblau, 2023-01-28:
Hallo gmbo,
zur Hardware. Ich habe einen selbstgebauten Batteriespeicher mit einem JBD BMS AP21S001. Als ESP32 ein DEV Board.
Auslesen mit der JBD App geht einwandfrei. Als Anzeige verwende ich den Seriellen Monitor des ESP.
Ich habe inzwischen einige Sketche gefunden die auch die Ausgabe auch über MQTT (Json) machen. Da sie alle auf dem Grundgerüst von Kolinz-cz beruhen habe ich bei allen den selben Fehler. Meiner laienhaften Meinung nach ist die PackBasicInfo leer durch fehlerhafte oder anders organisierten Daten. Also könnte es durchaus am BMS liegen. Produkt.Datum ist 10/22 also recht neu.  
Gruß
Wolfgang

gmbo, 2023-01-28:
Habe den Punkt für die Telegramme gefunden ist auch schon eingebaut muß nur herauskommentiert werden.
BMS_process_data.ino Zeile 245  hexDump(data,dataSize);
dann bekommst du als Ausgabe den Datensatz angezeigt.

Die scheinen wirklich etwas geändert zu haben, ich kann mit der App meinen Liontron-Akku nicht mehr auslesen.
Lies die Daten mal aus. 
Dann wäre es noch gut die Beschreibung zu finden ob die inzwischen anders ist.

bigmacblau, 2023-02-01:
Ich kann nichts finden :-( 
Ich habe inzwischen einen Sketch gefunden der speziell auf den JBD BMS AP21S001 angepasst ist. Hier erfolgt die Kommunikation allerdings über einen ESP8266 und UART Schnittstelle und platformio. Die Sketche sind doch zu unterschiedlich als das ich da Unterschiede fände. Evtl. DU ? Ist hier auf github.  [https://github.com/d4rken/jbd-bms-mqtt](url)
Gruß
Wolfgang

bigmacblau, 2023-02-02:
Den Hexdump habe ich aktiviert, wie kann ich den Speichern um ihn hier einzufügen? 
Was ich rausgefunden habe, der Unterschied ist ein zweiter Temp.Sensor.  Dadurch ist die Datenlänge ist 45 Byte. Der jbd-bms-mqtt Sketch funktioniert. !!!!
Die Hardware habe ich über Ali direkt, Die Kapazität des Akkus über die Handy App auf Referenz gesetzt bei vollständig geladenem Akku. 
Vielen Dank noch einmal für Deine Hilfe. 
Gruß
Wolfgang

gmbo, 2023-02-02:
Im serielle Monitor den autoscroll abschalten und nach dem Markieren des Textes strg c.
der ist dann in der Zwischenablage.
Hier dann als code einfügen.
Ich hatte mir schon gedacht, dass die Länge anders ist und du `  if (dataLen != 0x1D) { return false; } ` geändert hattest.

Das mit der Länge und der Checksumme ist im JBD-sketch besser gelöst. Dort wird die Länge im Telegramm abgefragt und damit gearbeitet.

IHateNameStealers, 2024-06-11:
2024 now, and it looks like I am having the exact same problem with one of my newer BMS'ses.
The older versions are working fine!
Did any of you guys managed to get this "invalid packet received" on the "printBasicInfo();" request?

2024 jetzt, und es sieht so aus, als hätte ich genau das gleiche Problem mit einem meiner neueren BMS's.
Die älteren Versionen funktionieren einwandfrei!
Hat es jemand von euch geschafft, dieses „ungültige Paket empfangen“ auf die „printBasicInfo();“ Anfrage zu bekommen?

Translated with [www.DeepL.com/Translator](http://www.deepl.com/Translator) (free version)

gmbo, 2024-06-11:
Was hast du für einen Type des BSM oder Akku? 
Ist es bei dir auch so, dass der Basic Record gar nicht empfangen wird?
Hast du mal Checksumtest ausgeschaltet und HEXdump aktiviert?

What type of BSM or battery do you have? 
Is it also the case with you that the basic record is not received at all?
Have you switched off Checksumtest and activated HEXdump?

IHateNameStealers, 2024-06-11:
Hello Gisbert,

I have two 12S1P battery's  a 12S9P and a 15S1P, and all battery's are DIY'ed, and are using te Overkill Solar (XiaoXiang) BMS's
One of the 12S1P BMS'es is giving me this "invalid packet received" error, however when I use Neil Shepherd (https://github.com/neilsheps/overkill-xiaoxiang-jbd-bms-ble-reader) his code, all works fine!
Unfortunately I really want to run this code on an ESP32.
Here is a dump including some HEX (hope it's enough)

`Forming a connection to a5:c2:37:15:41:73
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
The characteristic value was: 
We are now connected to the BLE Server.
HEX data:
0xdd, 0x4, 0x0, 0x18, 0xe, 0x40, 0xe, 0x42, 0xe, 0x43, 0xe, 0x3e, 0xe, 0x42, 0xe, 0x3f, 0xe, 0x40, 0xe, 0x3e, 
HEX data:
0xe, 0x42, 0xe, 0x42, 0xe, 0x41, 0xe, 0x3f, 0xfc, 0x3a, 0x77, 
HEX data:
0xdd, 0x3, 0x0, 0x24, 0x11, 0x1a, 0x0, 0x0, 0x8, 0xf8, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
HEX data:
0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x93, 0xb, 0x84, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf8, 0x0, 0x0, 
HEX data:
0xf9, 0xe1, 0x77, 
Invalid packer received
HEX data:
0xdd, 0x4, 0x0, 0x18, 0xe, 0x40, 0xe, 0x42, 0xe, 0x42, 0xe, 0x3e, 0xe, 0x42, 0xe, 0x3f, 0xe, 0x40, 0xe, 0x3e, 
HEX data:
0xe, 0x43, 0xe, 0x42, 0xe, 0x41, 0xe, 0x3f, 0xfc, 0x3a, 0x77, 
Total voltage: 0.000000
Amps: 0.000000
CapacityRemainAh: 0.000000
CapacityRemainPercent: 0
Temp1: 0.000000
Temp2: 0.000000
Balance Code Low: 0x0
Balance Code High: 0x0
Mosfet Status: 0x0
Number of cells: 12
Cell no. 1   3.648000
Cell no. 2   3.650000
Cell no. 3   3.650000
Cell no. 4   3.646000
Cell no. 5   3.650000
Cell no. 6   3.647000
Cell no. 7   3.648000
Cell no. 8   3.646000
Cell no. 9   3.651000
Cell no. 10   3.650000
Cell no. 11   3.649000
Cell no. 12   3.647000
Max cell volt: 3.651000
Min cell volt: 3.646000
Difference cell volt: 0.005000
Average cell volt: 3.648000
Median cell volt: 3.649000

HEX data:
0xdd, 0x3, 0x0, 0x24, 0x11, 0x1a, 0x0, 0x0, 0x8, 0xf8, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
HEX data:
0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x93, 0xb, 0x84, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf8, 0x0, 0x0, 
HEX data:
0xf9, 0xe1, 0x77, 
Invalid packer received
HEX data:
0xdd, 0x4, 0x0, 0x18, 0xe, 0x40, 0xe, 0x42, 0xe, 0x42, 0xe, 0x3e, 0xe, 0x42, 0xe, 0x3f, 0xe, 0x40, 0xe, 0x3d, 
HEX data:
0xe, 0x43, 0xe, 0x42, 0xe, 0x41, 0xe, 0x3f, 0xfc, 0x3b, 0x77, 
HEX data:
0xdd, 0x3, 0x0, 0x24, 0x11, 0x1a, 0x0, 0x0, 0x8, 0xf8, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
HEX data:
0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x93, 0xb, 0x84, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf8, 0x0, 0x0, 
HEX data:
0xf9, 0xe1, 0x77, 
Invalid packer received
HEX data:
0xdd, 0x4, 0x0, 0x18, 0xe, 0x40, 0xe, 0x42, 0xe, 0x42, 0xe, 0x3e, 0xe, 0x42, 0xe, 0x3f, 0xe, 0x40, 0xe, 0x3d, 
HEX data:
0xe, 0x42, 0xe, 0x42, 0xe, 0x41, 0xe, 0x3f, 0xfc, 0x3c, 0x77, 
HEX data:
0xdd, 0x3, 0x0, 0x24, 0x11, 0x1a, 0x0, 0x0, 0x8, 0xf8, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
HEX data:
0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x93, 0xb, 0x84, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf8, 0x0, 0x0, 
HEX data:
0xf9, 0xe1, 0x77, 
Invalid packer received
HEX data:
0xdd, 0x4, 0x0, 0x18, 0xe, 0x40, 0xe, 0x42, 0xe, 0x42, 0xe, 0x3e, 0xe, 0x42, 0xe, 0x3f, 0xe, 0x40, 0xe, 0x3d, 
HEX data:
0xe, 0x43, 0xe, 0x42, 0xe, 0x42, 0xe, 0x3f, 0xfc, 0x3a, 0x77, 
HEX data:
0xdd, 0x3, 0x0, 0x24, 0x11, 0x1a, 0x0, 0x0, 0x8, 0xf8, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
HEX data:
0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x93, 0xb, 0x84, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf8, 0x0, 0x0, 
HEX data:
0xf9, 0xe1, 0x77, 
Invalid packer received
HEX data:
0xdd, 0x4, 0x0, 0x18, 0xe, 0x40, 0xe, 0x42, 0xe, 0x42, 0xe, 0x3e, 0xe, 0x42, 0xe, 0x3f, 0xe, 0x40, 0xe, 0x3d, 
HEX data:
0xe, 0x42, 0xe, 0x43, 0xe, 0x41, 0xe, 0x3f, 0xfc, 0x3b, 0x77, 
HEX data:
0xdd, 0x3, 0x0, 0x24, 0x11, 0x1a, 0x0, 0x0, 0x8, 0xf8, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
HEX data:
0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x93, 0xb, 0x84, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf8, 0x0, 0x0, 
HEX data:
0xf9, 0xe1, 0x77, 
Invalid packer received`

IHateNameStealers, 2024-06-12:
I have enabled some more debugging options. Maybe it will help you debug the problem a bit more, as I have no clue how to fix it.

Ich habe einige weitere Debugging-Optionen aktiviert. Vielleicht hilft Ihnen das bei der Fehlersuche weiter, denn ich habe keine Ahnung, wie man das Problem beheben kann.

Translated with [www.DeepL.com/Translator](http://www.deepl.com/Translator) (free version)Translated with [www.DeepL.com/Translator](http://www.deepl.com/Translator) (free version)

> Forming a connection to a5:c2:37:15:41:73
>  - Created client
>  - Connected to server
>  - Found our service
>  - Found our characteristic
> The characteristic value was: 
> We are now connected to the BLE Server.
> HEX data:
> 0xdd, 0x3, 0x0, 0x24, 0x11, 0x19, 0x0, 0x0, 0x8, 0xf6, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
> HEX data:
> 0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x7e, 0xb, 0x72, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf6, 0x0, 0x0, 
> HEX data:
> 0xfa, 0xd, 0x77, 
> checksum: d5
> checksum v2: 2b
> Packet is not valid
> Expected value: 5
> Invalid packet received
> HEX data:
> 0xdd, 0x4, 0x0, 0x18, 0xe, 0x40, 0xe, 0x41, 0xe, 0x42, 0xe, 0x3d, 0xe, 0x42, 0xe, 0x3f, 0xe, 0x3f, 0xe, 0x3d, 
> HEX data:
> 0xe, 0x42, 0xe, 0x42, 0xe, 0x41, 0xe, 0x3e, 0xfc, 0x40, 0x77, 
> checksum: c0
> checksum v2: 40
> Packet is valid
> Total voltage: 0.000000
> Amps: 0.000000
> CapacityRemainAh: 0.000000
> CapacityRemainPercent: 0
> Temp1: 0.000000
> Temp2: 0.000000
> Max cell volt: 3.650000
> Min cell volt: 3.645000
> HEX data:
> 0xdd, 0x3, 0x0, 0x24, 0x11, 0x19, 0x0, 0x0, 0x8, 0xf6, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
> HEX data:
> 0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x7e, 0xb, 0x72, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf6, 0x0, 0x0, 
> HEX data:
> 0xfa, 0xd, 0x77, 
> checksum: d5
> checksum v2: 2b
> Packet is not valid
> Expected value: 5
> Invalid packet received
> HEX data:
> 0xdd, 0x4, 0x0, 0x18, 0xe, 0x40, 0xe, 0x41, 0xe, 0x42, 0xe, 0x3d, 0xe, 0x42, 0xe, 0x3f, 0xe, 0x3f, 0xe, 0x3d, 
> HEX data:
> 0xe, 0x42, 0xe, 0x42, 0xe, 0x41, 0xe, 0x3e, 0xfc, 0x40, 0x77, 
> checksum: c0
> checksum v2: 40
> Packet is valid
> Total voltage: 0.000000
> Amps: 0.000000
> CapacityRemainAh: 0.000000
> CapacityRemainPercent: 0
> Temp1: 0.000000
> Temp2: 0.000000
> Max cell volt: 3.650000
> Min cell volt: 3.645000
> HEX data:
> 0xdd, 0x3, 0x0, 0x24, 0x11, 0x19, 0x0, 0x0, 0x8, 0xf6, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
> HEX data:
> 0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x7e, 0xb, 0x72, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf6, 0x0, 0x0, 
> HEX data:
[... 92 lines cut ...]
> Amps: 0.000000
> CapacityRemainAh: 0.000000
> CapacityRemainPercent: 0
> Temp1: 0.000000
> Temp2: 0.000000
> Max cell volt: 3.650000
> Min cell volt: 3.645000
> HEX data:
> 0xdd, 0x3, 0x0, 0x24, 0x11, 0x19, 0x0, 0x0, 0x8, 0xf6, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
> HEX data:
> 0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x7e, 0xb, 0x72, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf6, 0x0, 0x0, 
> HEX data:
> 0xfa, 0xd, 0x77, 
> checksum: d5
> checksum v2: 2b
> Packet is not valid
> Expected value: 5
> Invalid packet received
> HEX data:
> 0xdd, 0x4, 0x0, 0x18, 0xe, 0x40, 0xe, 0x41, 0xe, 0x42, 0xe, 0x3d, 0xe, 0x42, 0xe, 0x3f, 0xe, 0x3f, 0xe, 0x3d, 
> HEX data:
> 0xe, 0x42, 0xe, 0x42, 0xe, 0x40, 0xe, 0x3f, 0xfc, 0x40, 0x77, 
> checksum: c0
> checksum v2: 40
> Packet is valid

IHateNameStealers, 2024-06-12:
Okay. After your suggested dataLen != 0x24 and disabling the if (isValid != true) { ... } check, it's kind of working.
Total voltage, Amps and CapacityRemainAh are showing correct values.
CapacityRemainPercent, Temp1 and Temp2 are showing wrong values.

Okay. Nach Ihrem vorgeschlagenen dataLen != 0x24 und Deaktivieren der if (isValid != true) { ... } Prüfung, funktioniert es irgendwie.
Gesamtspannung, Ampere und CapacityRemainAh zeigen korrekte Werte an.
CapacityRemainPercent, Temp1 und Temp2 zeigen falsche Werte an.

> Total voltage: 43.770000
> Amps: 0.000000
> CapacityRemainAh: 22.980000
> CapacityRemainPercent: 66
> Temp1: 6280.500000
> Wrong - Temp2: 6280.500000
> 
>

IHateNameStealers, 2024-06-12:
I found this. Maybe it's of some help?

Ich habe das hier gefunden. Vielleicht ist es eine Hilfe?

https://www.dropbox.com/s/03vfqklw97hziqr/%E9%80%9A%E7%94%A8%E5%8D%8F%E8%AE%AE%20V2%20%28%E6%94%AF%E6%8C%8130%E4%B8%B2%29%28Engrish%29.xlsx?dl=0

gmbo, 2024-06-12:
Hi,
I also found the documentation JBD protocol english this morning, but it is still the same as the one from the page 
[https://www.lithiumbatterypcb.com/wp-content/uploads/2023/05/RS485-UART-RS232-Communication-protocol.pdf](https://www.lithiumbatterypcb.com/wp-content/uploads/2023/05/RS485-UART-RS232-Communication-protocol.pdf)
I had hoped that changing the bytes alone would work. 
Where did you comment out the **if (isValid != true) { ... } check**, commented out?
Actually, the values should be correct when the routine runs. I'll have to look more closely to see where something can go wrong.  I am afraid that the length of the short telegram is taken when calculating the checksum. I just haven't worked on the source code for ages. My system runs in the motorhome on a Liontron 200AH and transmits the data home via LoraWAN. The last thing I had to change was to exclude coupling to an external battery. That has happened before.  
Looking at your dump you should have a 50AH battery and temperature 1 is 21.8°C (0xb85 = 2949 -2731 = 218) and temperature 2 is 20.4 °C.
There must be something wrong in the counting method.  I used the table for this and it fits.

I'm like you with the language. My English isn't good enough either and I also use DeepL in the free version. But they actually do a pretty good job.

Translated with DeepL.com (free version)

Hi,
die Doku JBD protokoll engnglisch hatte ich heute morgen auch gefunden, ist aber noch die gleiche wie die von der Seite 
[https://www.lithiumbatterypcb.com/wp-content/uploads/2023/05/RS485-UART-RS232-Communication-protocol.pdf](https://www.lithiumbatterypcb.com/wp-content/uploads/2023/05/RS485-UART-RS232-Communication-protocol.pdf)
Ich hatte gehofft, dass allein Änderung der Bytes klappt. 
Wo hast du den  **if (isValid != true) { ... } check**, auskommentiert?
Eigentlich sollten die Werte richtig sein wenn die Routine durchläuft. Ich muß mal genauer suchen wo da etwas schief laufen kann.  Ich habe die Befürchtung, dass beim berechnen der Cecksumme die Länge des kurzen Telegramms genommen wird. War nur jetzt ewig nicht mehr am Quellcode. Mein System läuft im Wohnmobil an einer Liontron 200AH und überträgt die Daten per LoraWAN nach Hause. Das letzte was ich ändern mußte war das Koppeln an eine fremde Batterie auszuschließen. Das passierte schon mal.  
Wenn ich deinen Dump ansehe solltest du eine 50AH Batterie haben und die Temperatur 1 liegt bei 21,8°C (0xb85 = 2949 -2731 = 218)  und Temperatur 2 bei 20,4 °C.
Da müsste etwas in der Zählweise  falsch sein.  Ich hatte dafür extra die Tabelle eingesetzt und das passt.

Mit der Sprache geht es mir wie dir. Mein Englisch reicht auch nicht aus und auch ich benutze DeepL in der Freeversion. Die machen das aber eigentlich recht gut.

IHateNameStealers, 2024-06-12:
Once again, thank you for your help!

The reason I removed the if **(isValid != true) { ... }** is because I could not read the data from the "old working" BMS anymore. I am using the ESP32 to read several BMS units. As mentioned above, one (the new one) is giving me a hard time. So, yes, **dataLen != 0x24** works with the new BMS, but not with the old one. 

I will have a look at Neil Sheps' code (https://github.com/neilsheps/overkill-xiaoxiang-jbd-bms-ble-reader) and see if I can figure out why his code works fine with my old and new BMS.

Nochmals vielen Dank für Ihre Hilfe!

Der Grund, warum ich das if **(isValid != true) { ... }** entfernt habe, ist, dass ich die Daten des "alten" BMS nicht mehr auslesen konnte. Ich verwende den ESP32, um mehrere BMS-Geräte auszulesen. Wie oben erwähnt, macht eine (die neue) mir Schwierigkeiten. Also, ja, **dataLen != 0x24** funktioniert mit dem neuen BMS, aber nicht mit dem alten. 

Ich werde mir den Code von Neil Sheps (https://github.com/neilsheps/overkill-xiaoxiang-jbd-bms-ble-reader) ansehen und sehen, ob ich herausfinden kann, warum sein Code mit meinem alten und neuen BMS funktioniert.

gmbo, 2024-06-12:
If I'm not mistaken, it omits the query if dataLen completely. Looking at this, I had thought to see if it could be completely omitted and a telegram only determined by type, i.e. 3 Basic and then the validity according to the byte length and data. 
[/neilsheps/overkill-xiaoxiang-jbd-bms-ble-reader](https://github.com/neilsheps/overkill-xiaoxiang-jbd-bms-ble-reader reads in any number of temperature values in a for loop, for example. The values are only output and not saved in a variable.
Something similar is done in the 4 telegram, where the cells are also saved by number.

Translated with DeepL.com (free version)

Wenn mich nicht alles täuscht läßt der die Abfrage if dataLen  ganz weg. Wenn ich mir das ansehe hatte ich gedacht mal zu schauen ob man da komplett drauf verzichten kann und ein Telegramm nur nach Type also 3 Basic und dann die Gültigkeit nach dem Byte länge und Daten bestimmt. 
[/neilsheps/overkill-xiaoxiang-jbd-bms-ble-reader](https://github.com/neilsheps/overkill-xiaoxiang-jbd-bms-ble-reader liest zum Beispiel beliebig viele Temperaturwerte in einer for Schleife ein. Es werden ja die Werte nur ausgegeben und nicht in einer Variablen gespeichert.
Im 4er Telegramm wird ja auch so etwas gemacht, da werden die Zellen ja auch nach Anzahl gespeichert.

Es wäre super wenn jemand den Grund für die Erweiterung um 9 Bytes kennt, dann hätte man eine Chance das zu erweitern.

gmbo, 2024-06-13:
Du bekommst den Checksum error immer wenn du prüfst,  Das wird am Buffer liegen setze mal 
`    static byte bmsPacketBuff[40];`
auf 
`    static byte bmsPacketBuff[50];`
dann müsste das behoben sein.
und kommentiere die Längenabfrage aus
```
    // Expected data len
 /*   if (dataLen != 0x1B)
    {
        return false;
    }
*/ 

```
dann sollte alt und neu gehen.

IHateNameStealers, 2024-06-13:
I have changed the **bmsPacketBuff** size from **40** to **50** and commented out the check with /**/ Expected data len**. Unfortunately, I am still experiencing issues with **CapacityRemainPercent**, **Temp1**, and **Temp2**. Additionally, the **isValid check (if (isValid != true))** is still resulting in an "**Invalid packet received**" error.

Ich habe die **bmsPacketBuff** Größe von **40** auf **50** geändert und die Prüfung mit /**/ Expected data len** auskommentiert. Leider habe ich immer noch Probleme mit **CapacityRemainPercent**, **Temp1** und **Temp2**. Außerdem führt die **isValid-Prüfung (if (isValid != true))** immer noch zu einem Fehler "**Ungültiges Paket empfangen**".

gmbo, 2024-06-13:
I have specified the buffer because telegram 3 is now 43 bytes long.
There is another 2. 
set packetbuff[40] to at least 50.
If this does not help 
you could enter the following in Smart-BMS-Bluetooth-ESP32.ino at the top 

```
#define TRACE commSerial.println(__FUNCTION__)
//#define TRACE

```

and make another dump, 

Den Buffer habe ich angegeben, da das Telegramm 3  inzwischen 43 Bytes lang ist.
es gibt noch einen 2. 
packetbuff[40]  den auch mal auf mindestens 50 setzen.
Wenn das nicht hilft 
könntest du in   Smart-BMS-Bluetooth-ESP32.ino mal ganz oben 

```
#define TRACE commSerial.println(__FUNCTION__)
//#define TRACE

```

ändern und noch mal einen Dump machen.

IHateNameStealers, 2024-06-13:
I have done all modifications as suggested above, but still with no luck. Please see attached HEX dump.

Ich habe alle Änderungen wie oben vorgeschlagen, aber immer noch kein Glück. Bitte siehe beigefügten HEX-Dump.

> connectToServer
> Forming a connection to a5:c2:37:15:41:73
>  - Created client
>  - Connected to server
>  - Found our service
>  - Found our characteristic
> The characteristic value was: 
> We are now connected to the BLE Server.
> bmsGetInfo4
> sendCommand
> Request info4 sent
> bmsGetInfo3
> sendCommand
> Request info3 sent
> notifyCallback
> hexDump
> HEX data:
> 0xdd, 0x3, 0x0, 0x24, 0x11, 0x18, 0x0, 0x0, 0x8, 0xf0, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
> bleCollectPacket
> hexDump
> HEX data:
> 0xdd, 0x3, 0x0, 0x24, 0x11, 0x18, 0x0, 0x0, 0x8, 0xf0, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
> notifyCallback
> hexDump
> HEX data:
> 0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x8c, 0xb, 0x7e, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf0, 0x0, 0x0, 
> bleCollectPacket
> hexDump
> HEX data:
> 0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x8c, 0xb, 0x7e, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf0, 0x0, 0x0, 
> notifyCallback
> hexDump
> HEX data:
> 0xfa, 0x0, 0x77, 
> bleCollectPacket
> hexDump
> HEX data:
> 0xfa, 0x0, 0x77, 
> bmsProcessPacket
> isPacketValid
> checksum: 3a
> checksum v2: c6
> Packet is not valid
> Expected value: 1c
> processBasicInfo
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> printBasicInfo
> Total voltage: 43.759998
> Amps: 0.000000
> CapacityRemainAh: 22.879999
> CapacityRemainPercent: 63
> Temp1: 1364.900024
> Temp2: 955.500000
> Max cell volt: 0.000000
> Min cell volt: 0.000000
> bmsGetInfo4
> sendCommand
> Request info4 sent
> notifyCallback
> hexDump
> HEX data:
> 0xdd, 0x4, 0x0, 0x18, 0xe, 0x3f, 0xe, 0x41, 0xe, 0x41, 0xe, 0x3d, 0xe, 0x41, 0xe, 0x3e, 0xe, 0x3f, 0xe, 0x3c, 
> bleCollectPacket
> hexDump
> HEX data:
> 0xdd, 0x4, 0x0, 0x18, 0xe, 0x3f, 0xe, 0x41, 0xe, 0x41, 0xe, 0x3d, 0xe, 0x41, 0xe, 0x3e, 0xe, 0x3f, 0xe, 0x3c, 
> notifyCallback
> hexDump
> HEX data:
> 0xe, 0x41, 0xe, 0x41, 0xe, 0x40, 0xe, 0x3e, 0xfc, 0x48, 0x77, 
> bleCollectPacket
> hexDump
> HEX data:
> 0xe, 0x41, 0xe, 0x41, 0xe, 0x40, 0xe, 0x3e, 0xfc, 0x48, 0x77, 
> bmsProcessPacket
> isPacketValid
> checksum: b8
> checksum v2: 48
> Packet is valid
> processCellInfo
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> two_ints_into16
> printBasicInfo
> Total voltage: 43.759998
> Amps: 0.000000
> CapacityRemainAh: 22.879999
> CapacityRemainPercent: 63
> Temp1: 1364.900024
> Temp2: 955.500000
> Max cell volt: 3.649000
> Min cell volt: 3.644000
> bmsGetInfo3
> sendCommand
> Request info3 sent
> notifyCallback
> hexDump
> HEX data:
> 0xdd, 0x3, 0x0, 0x24, 0x11, 0x18, 0x0, 0x0, 0x8, 0xf0, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
> bleCollectPacket
> hexDump
> HEX data:
> 0xdd, 0x3, 0x0, 0x24, 0x11, 0x18, 0x0, 0x0, 0x8, 0xf0, 0x13, 0x88, 0x0, 0x0, 0x30, 0xc7, 0x0, 0x0, 0x0, 0x0, 
> notifyCallback
> hexDump
> HEX data:
> 0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x8c, 0xb, 0x7e, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf0, 0x0, 0x0, 
> bleCollectPacket
> hexDump
> HEX data:
> 0x10, 0x0, 0x29, 0x2e, 0x1, 0xc, 0x2, 0xb, 0x8c, 0xb, 0x7e, 0x0, 0x0, 0x0, 0x13, 0x88, 0x8, 0xf0, 0x0, 0x0, 
> notifyCallback
> hexDump
> HEX data:
> 0xfa, 0x0, 0x77, 
> bleCollectPacket
> hexDump
> HEX data:
> 0xfa, 0x0, 0x77, 
> bmsProcessPacket
> isPacketValid
> checksum: 3a
> checksum v2: c6
> Packet is not valid
> Expected value: 1c

gmbo, 2024-06-13:
Hab direkt noch keine Idee. Liest das Programm denn jetzt die alten Telegramme?
Ich muß doch mal schauen,dass ich eine Batterie in Reichweite habe,.

IHateNameStealers, 2024-06-14:
I am lost too, but will not give up on this project.
I also appreciate all your help trying to fix this problem.

Ich bin auch ratlos, aber ich werde dieses Projekt nicht aufgeben.
Ich danke Ihnen auch für Ihre Hilfe bei der Lösung dieses Problems.

gmbo, 2024-06-14:
I didn't want to give up either. I had another look at Neil Shepherd's version, which even works with 
`#define MAX_BMS_DATA_CAPACITY 100`
But I still have an idea.
The old telegram does work.
In the function 
`bool bleCollectPacket(char *data, uint32_t dataSize)`
it could be that 2 packets are assumed. The old telegram does not send any more. 

Aufgeben wollte ich auch nicht. Hatte mir die Version von Neil Shepherd noch mal angesehen , der arbeitet sogar mit 
`#define MAX_BMS_DATA_CAPACITY		100`
Habe aber doch gerade noch eine Idee.
Das alte Telegramm funktioniert doch.
In der Funktion 
`bool bleCollectPacket(char *data, uint32_t dataSize)`
könnte es sein, dass von 2 Paketen ausgegangen wird. mehr sendet das alte Telegramm nicht. 
Das neue sendet ein 3. Paket und das könnte alles durcheinander bringen.

gmbo, 2024-06-15:
I have edited the function a little, others can certainly do it 100 times better, but it also puts together long telegrams.
The routine would have to be completely rewritten. But if need be, it works like this too.
The isvalid check works with it.
If you now allow 0x1b and 0x24 in the query for the length, you can also make an individual display for the old and new battery type with switch case.

Habe mal die Funktion etwas bearbeitet, das können andere bestimmt 100 mal besser , aber so setzt es auch lange Telegramme zusammen.
Man müsste die Routine mal komplett umschreiben. Aber zur Not geht se auch so.
Die Prüfung isvalid funktioniert damit.
Wenn jetzt in der Abfrage nach der Länge 0x1b und 0x24 zuläßt kann man auch eine individuelle Anzeige für den alten und neuen Batterietyp mit switch case  machen.

```
switch(dataLen ) {
	case 0x1B: printf("Alter BMS-Typ\n"); break;
	case 0x24: printf("Neuer BMS-Typ\n"); break;
	default: printf("falscher Typ\n");  return false;
}

```

```

bool bleCollectPacket(char *data, uint32_t dataSize) // reconstruct packet from BLE incomming data, called by notifyCallback function
{
  TRACE;
  static uint8_t packetstate = 0; //0 - empty, 1 - first half of packet received, 2- second half of packet received
  static uint8_t packetbuff[100] = {0x0};
  static uint32_t previousDataSize = 0;

  static uint32_t newDataSize = 0;
  bool retVal = false;
//  hexDump(data, dataSize);
  // commSerial.printf("Telegrammpart: %d - pre: %d -  new: %d - dataSize: %d \n", packetstate, previousDataSize, newDataSize, dataSize);

  if (data[0] == 0xdd && packetstate == 0) // probably got 1st half of packet
  {
    packetstate = 1;
//    commSerial.printf("in: %d - pre: %d -  new: %d - dataSize: %d \n", packetstate, previousDataSize, newDataSize, dataSize);
    previousDataSize = dataSize;
//    commSerial.printf("0x%02x 0x%02x 0x%02x 0x%02x -- %d \n", data[0], data[1], data[2], data[3], dataSize);

    for (uint8_t i = 0; i < dataSize; i++)
    {
      packetbuff[i] = data[i];
    }
    retVal = false;
  }

  if (data[dataSize - 1] == 0x77 && packetstate >= 1) //probably got 2nd half of the packet
  {
    packetstate = 3;
 //   commSerial.printf("in last: %d - pre: %d -  new: %d - dataSize: %d \n", packetstate, previousDataSize, newDataSize, dataSize);
 //   commSerial.printf("0x%02x 0x%02x 0x%02x -- %d \n", data[0], data[1], data[2], dataSize);

    for (uint8_t i = 0; i < dataSize; i++)
    {
      packetbuff[i + previousDataSize] = data[i];
    }
    retVal = false;
  }
  else
  {
    if (data[0] != 0xDD ) //
    {

      packetstate = 2;
//      commSerial.printf("in Mitte: %d - pre: %d -  new: %d - dataSize: %d \n", packetstate, previousDataSize, newDataSize, dataSize);
//      commSerial.printf("0x%02x 0x%02x 0x%02x -- %d \n", data[0], data[1], data[2], dataSize);

      newDataSize = previousDataSize;
      for (uint8_t i = 0; i < previousDataSize; i++)
      {
        packetbuff[i + previousDataSize] = data[i];
      }
      previousDataSize = newDataSize + dataSize ;
      retVal = false;
    }
  }

  if (packetstate == 3) //got full packet
  {
    uint8_t packet[dataSize + previousDataSize];
    memcpy(packet, packetbuff, dataSize + previousDataSize);
/*    commSerial.printf("Telegrammende: %d - pre: %d -  new: %d - dataSize: %d \n", packetstate, previousDataSize, newDataSize, dataSize);
    for (int i = 0; i < 43; i++)
    {
      commSerial.printf("0x%x, ", packet[i]);
    }
    commSerial.println();
*/
    bmsProcessPacket(packet); //pass pointer to retrieved packet to processing function
    packetstate = 0;
    newDataSize = 0;
    previousDataSize = 0;
    retVal = true;
  }
  return retVal;
}

```

IHateNameStealers, 2024-06-15:
Hallo Gisbert,

JA. Beide Funktionen funktionieren jetzt gut mit deinem neuen Code! Herzlichen Dank dafür!

Zu deiner Information vor deiner Korrektur. 
BMS mit Firmware: 2.2 funktioniert.
BMS mit Firmware: 2.5 NICHT funktionierend
BMS mit Firmware: 2.9 NICHT funktionierend
BMS mit Firmware: 3.8 NICHT funktionierend

Nach Ihrer Korrektur funktioniert der Code mit allen oben genannten Firmware-Versionen einwandfrei.
Gisbert. Du hast meinen Tag gerettet!

gmbo, 2024-06-15:
I didn't realize there were so many firmware versions. all newer versions provide the same length of user data.

Ich wußte gar nicht, dass es so viele Firmwareversionen gibt. liefern alle neueren Versionen die selbe Länge der Nutzdaten.

IHateNameStealers, 2024-06-16:
For what I can see, all the versions after 2.2 have the same length of user data.

Soweit ich sehen kann, haben alle Versionen nach 2.2 die gleiche Länge der Benutzerdaten.

gmbo, 2024-06-16:
I have found the description for the 9 additional bytes.

Data |Format |Unit 	
--------|--------|--------
Humidity 	|1 byte 	|1 percent 		
Alarm Status 	|2 bytes 	|not used normally 		
Full charge capacity 	|2 bytes 	|10 mAH 		
Remaining capacity 	|2 bytes 	|10 mAH 		
Balance current 	|2 bytes 	|1 mA

Ich habe die Beschreibung für die 9 Zusätzlichen Bytes gefunden.
Daten |Format |Einheit	
--------|--------|--------
Luftfeuchtigkeit |1 Byte |1 Prozent 		
Alarmstatus |2 Byte |normalerweise nicht verwendet 		
Volle Ladekapazität |2 Byte |10 mAH 		
Verbleibende Kapazität |2 Byte |10 mAH 		
Ausgleichsstrom |2 Byte |1 mA

IHateNameStealers, 2024-06-17:
Good find Gisbert. Kudos to you!

Ein guter Fund, Gisbert. Ein großes Lob an Sie!

gmbo, 2024-06-17:
[github.com/FurTrader/OverkillSolarBMS](https://github.com/FurTrader/OverkillSolarBMS)
is the most comprehensive description and collection of tools I have found so far.

ist die umfangreichste Beschreibung und Toolsammung die ich bisher fand.

gmbo, 2024-06-17:
I have extended the basic telegram and inserted the query for new BMS. Is now inserted by kolins-cz in the masterbranch.

Ich habe das Basictelegramm erweitert und die Abfrage auf neue BMS eingefügt. Ist jetzt von kolins-cz im masterbranch eingefügt.

IHateNameStealers, 2024-06-17:
I will test the masterbranch tomorrow, and will let you know the results.
Once again thank you.

Ich werde die Hauptverzweigung morgen testen und Ihnen die Ergebnisse mitteilen.
Ich danke Ihnen nochmals.

gmbo, 2024-06-19:
Sorry, I have a bug in mydatatypes.h. I am currently correcting it. 
Then it could not be translated in the Arduino IDE2 because of LCD.
I also changed that and added user setup files for the Arduitouch and TTgo_T-beam.
Try a pull request next.

Entschuldigung ich habe im mydatatypes.h einen Bug . den korregiere ich gerade. 
Dann lies es sich in der Arduino IDE2 nicht übersetzen, wegen LCD.
Habe das auch geändert und ich habe User Setup-Dateien für den Arduitouch und TTgo_T-beam hinzugefügt.
Versuche gleich einen Pull Request.

gmbo, 2024-06-28:
This should now be translatable. The pull request has been entered in the master branch.

So jetzt sollte das übersetzbar sein. Der Pull request ist im Master branch eingepflegt.

NeariX67, 2024-08-04:
Relying on the data length do determine whether the BMS has a new or old firmware is not safe. I just set up this project and noticed errors with "falscher typ" because i have more than two temperature sensors that directly impact the length of the message.
Maybe i'll do a PR

PR #14 (2024-08-28): remove data length check in processBasicInfo
This fixes incompatibility with bms that feature less or more than two temperature sensors. The current implementation only contained indexes for a BMS with two temperature sensors, so it failed if a different amount was present.

Should close #7
closed 2024-08-28 by kolins-cz