kolins-cz/Smart-BMS-Bluetooth-ESP32 #8: esp32 keeps restarting

anshilike, 2023-10-22:
Hello, I can connect to the bms with the ordinary esp32wroom development board and the ESP32 ARDUINO PACKAGE2.0.3 burning program, but there is an error and I cannot read the data, 
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
The characteristic value was: 

assert failed: heap_caps_free heap_caps.c:339 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace:0x40083639:0x3ffc8f500x4009254d:0x3ffc8f70 0x40097525:0x3ffc8f90 0x40083972:0x3ffc90c0 0x40097555:0x3ffc90e0 0x4017ed75:0x3ffc9100 0x4017fc1c:0x3ffc9120 0x400d276d:0x3ffc9140 0x400d3483:0x3ffc9190 0x400d351e:0x3ffc91d0 0x400df739:0x3ffc91f0 

and I can not compile with arduino package 1.0.6. May I ask is it my choice of development board or the choice of esp32 arduino package wrong or other problems? Hope to hear from you. Thank you.

kolins-cz, 2023-10-24:
Sorry, I have no idea. I do not work on this any more.

gmbo, 2023-10-24:
Hallo,
ich weiß nicht ob das hilft, aber esgab bei anderen Projekten leichte Schwierigkeiten bei der Umstellung von der Arduino-IDE nach 2.0.
Ich habe mehrere Abwandlungen mit der Arduino-IDE 1.8.19 am laufen auf verschiedenen ESP32 Boards. z.B. ESP32-DevKitC oder LILYGO T-Beam Meshtastic LORA32.
Es würde bestimmt Sinn machen es mal mit einer IDE 1.8.x zu testen.
Manchmal ist es günstig vor dem Aufspielen der Software mit dem 
esptool den Speicher im ESP zu löschen.
Unter Linux: esptool --port /dev/ttyUSB0 erase_flash
Das hat bei vielen Problemen geholfen.

Hello,
I don't know if this helps, but there were slight difficulties in other projects when switching from the Arduino IDE to 2.0.
I have several variations with the Arduino IDE 1.8.19 running on different ESP32 boards. e.g. ESP32-DevKitC or LILYGO T-Beam Meshtastic LORA32.
It would certainly make sense to test it with an IDE 1.8.x.
Sometimes it is convenient before installing the software with the 
esptool to clear the memory in the ESP.
Under Linux: esptool --port /dev/ttyUSB0 erase_flash
This has helped with many problems.

Translated with www.DeepL.com/Translator (free version)

anshilike, 2023-10-26:
> 你好, 他说:"我想,我想我们应该在这里工作。 1.8.19美国劳芬公司的ESP32板。Z.B.Esp32德福基特奥德利利戈t-波姆梅萨32。 这是一个新概念. 全人类主义者 用工具说明她的身份。 工具--端口/发展/Ttyusb0Erase_flash 海贝维伦问题专家格霍芬。
> 
> 你好, 我不知道这是否有帮助,但是当从ArdunoIDD转换为2.0时,其他项目中也存在一些轻微的困难。 我有几个不同的变化,在不同的ESP32板上运行的阿杜伊诺ID1.8.19。例如。ESP32-DEVITKIC或Lilygot-Beam的洛拉32。 当然,用一个1.8.X测试它是有意义的。 有时候在安装软件前使用 用电子工具来清除特别程序中的内存。 在Linux下:ESPBLU工具----端口/发展/Ttyusb0Erase_flash 这有助于解决许多问题。
> 
> 翻译为[ /翻译](http://www.DeepL.com/Translator) (免费版)

I burned the esp32 wroom32 development board into the micropython firmware and then wrote the arduino firmware into it. Can I clear the flash?However, the above situation still occurs. I am using arduino1.8.19 and esp32 arduino packge 2.0.3. Finally, would you like to ask if this problem will be solved if I use the ttgo development board? Thank you.

gmbo, 2023-10-27:
https://docs.espressif.com/projects/esptool/en/latest/esp32/
Mit dem Tool ist es möglich den Speicher zu löschen um dann mit der Arduino IDE ein Programm zu laden.
Der TTGO T-Beam ist nicht der einfachste Kandidat für das  Programm wird auch nur dann gebraucht wenn die Daten per LoRAWAN übertragen werden sollen.
Am einfachsten sind die Standard ESP Boards, es ist nur darauf zu achten in der IDE die richtigen Einstellungen für Speichergröße ( ich verwende Huge APP (3MB no OTA/1MB SPIFSS)) und PSRAM: "disabled" und aufpassen dass das richtige ESP32 oder ESP32S3 oder ESP32C3 eingestellt ist.

https://docs.espressif.com/projects/esptool/en/latest/esp32/
With the tool it is possible to clear the memory to load a program with the Arduino IDE.
The TTGO T-Beam is not the easiest candidate for the program and is only needed if the data is to be transferred via LoRAWAN.
The easiest way is to use the standard ESP boards, you just have to take care in the IDE to use the right settings for memory size (I use Huge APP (3MB no OTA/1MB SPIFSS)) and PSRAM: "disabled" and take care that the right ESP32 or ESP32S3 or ESP32C3 is set.

Translated with www.DeepL.com/Translator (free version)

TjadenWright, 2023-10-29:
I am having this same issue on Arduino 2.2.1 with Memory Size: Huge APP (3MB no OTA/1MB SPIFSS) and PSRAM: "disabled".

dnmatt, 2024-02-16:
I found a bug in the routine "connectToServer" in ble.ino. At the very end when you have connected successfully you have:

         BLE_client_connected = true;
}

Since the routine is a bool, a true/false needs to be returned. Should be:

         BLE_client_connected = true;
         return true;
}

Once I made that change the reboots went away.